paperdink/PaperdInk-Library #1: Calther Example Does Not Compile

heytcass, 2022-12-28:
Trying to flash Calther to my Paperd.ink Classic, and keep getting the following errors:

```
/home/tomcassady/Arduino/calther/calther.ino: In function 'void setup()':
/home/tomcassady/Arduino/calther/calther.ino:59:20: error: 'class PaperdinkUIDateClass' has no member named 'display_calender'
     Paperdink_Date.display_calender(Paperdink.epd, 125, 40); 
                    ^
/home/tomcassady/Arduino/calther/calther.ino:63:20: error: 'class PaperdinkUIDateClass' has no member named 'display_day_date_style2_center'
     Paperdink_Date.display_day_date_style2_center(Paperdink.epd, 0, 205, 110);
                    ^
/home/tomcassady/Arduino/calther/calther.ino:66:65: error: no matching function for call to 'PaperdinkUIWeatherClass::fetch_data(const char [16], const char [3], const char [33], const char [2])'
     if(Paperdink_Weather.fetch_data(CITY, COUNTRY, OWM_ID, UNITS) < 0){
                                                                 ^
In file included from /home/tomcassady/Arduino/libraries/Paperdink/src/Paperdink.h:19:0,
                 from /home/tomcassady/Arduino/calther/calther.ino:25:
/home/tomcassady/Arduino/libraries/Paperdink/src/ui/weather/weather.h:13:16: note: candidate: int8_t PaperdinkUIWeatherClass::fetch_data(const char*, const char*, const char*)
         int8_t fetch_data(const char* city, const char* country, const char* owm_api_id);
                ^
/home/tomcassady/Arduino/libraries/Paperdink/src/ui/weather/weather.h:13:16: note:   candidate expects 3 arguments, 4 provided
/home/tomcassady/Arduino/calther/calther.ino:72:23: error: 'class PaperdinkUIWeatherClass' has no member named 'display_weather_style1'
     Paperdink_Weather.display_weather_style1(Paperdink.epd, 5, 22);
                       ^
/home/tomcassady/Arduino/calther/calther.ino:73:23: error: 'class PaperdinkUIWeatherClass' has no member named 'display_weather_style2_center'
     Paperdink_Weather.display_weather_style2_center(Paperdink.epd, 0, 130, 110);
                       ^
/home/tomcassady/Arduino/calther/calther.ino:77:5: error: 'Paperdink_TodoListTodoist' was not declared in this scope
     Paperdink_TodoListTodoist.font_bold = &PAPERDINK_FONT_SML_BOLD;
     ^

exit status 1

Compilation error: 'class PaperdinkUIDateClass' has no member named 'display_calender'
```

Have tried in both Platformio and Arduino IDE with the same errors. All dependencies are installed.

dmikalova, 2022-12-29:
I got a similar issue with the Todoist example - it turns out that the PaperdInk Library v0.0.2 that has been released and uploaded Arduino has not been updated for the latest examples. I was able to compile them by downloading and zipping the repo and [following the Arduino docs](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries) for manually uploading a library. Note if you clone the repo you will need to remove the `.git` dir from the zip.

emilyboda, 2023-01-06:
Hi @dmikalova, I just want to make sure I'm reading your explanation correctly. Your issue was fixed by manually updating the PaperdInk Library from v0.0.1 to v0.0.2? I'm experiencing the same issue as you and the OP have but I already have Paperdink Library v0.0.2 installed (that is the latest one available in the Arduino Library Manager right now).

hakova, 2023-01-08:
Same error here. I noticed the typo and changed it to 'display_calend**a**r' with no avail. See the error messages below:

```
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/home/hakayova/Arduino/hakayova/calther/calther.ino: In function 'void setup()':
calther:59:20: error: 'class PaperdinkUIDateClass' has no member named 'display_calendar'
     Paperdink_Date.display_calendar(Paperdink.epd, 125, 40); 
                    ^
calther:63:20: error: 'class PaperdinkUIDateClass' has no member named 'display_day_date_style2_center'
     Paperdink_Date.display_day_date_style2_center(Paperdink.epd, 0, 205, 110);
                    ^
calther:66:65: error: no matching function for call to 'PaperdinkUIWeatherClass::fetch_data(const char [10], const char [14], const char [33], const char [2])'
     if(Paperdink_Weather.fetch_data(CITY, COUNTRY, OWM_ID, UNITS) < 0){
                                                                 ^
In file included from /home/hakayova/Arduino/libraries/Paperdink/src/Paperdink.h:19:0,
                 from /home/hakayova/Arduino/hakayova/calther/calther.ino:25:
/home/hakayova/Arduino/libraries/Paperdink/src/ui/weather/weather.h:13:16: note: candidate: int8_t PaperdinkUIWeatherClass::fetch_data(const char*, const char*, const char*)
         int8_t fetch_data(const char* city, const char* country, const char* owm_api_id);
                ^
/home/hakayova/Arduino/libraries/Paperdink/src/ui/weather/weather.h:13:16: note:   candidate expects 3 arguments, 4 provided
calther:72:23: error: 'class PaperdinkUIWeatherClass' has no member named 'display_weather_style1'
     Paperdink_Weather.display_weather_style1(Paperdink.epd, 5, 22);
                       ^
calther:73:23: error: 'class PaperdinkUIWeatherClass' has no member named 'display_weather_style2_center'
     Paperdink_Weather.display_weather_style2_center(Paperdink.epd, 0, 130, 110);
                       ^
calther:89:29: error: 'class PaperdinkUITodoListTaigaClass' has no member named 'primary_color'
     Paperdink_TodoListTaiga.primary_color = GxEPD_WHITE;
                             ^
calther:94:29: error: 'class PaperdinkUITodoListTaigaClass' has no member named 'display_list_style1'
     Paperdink_TodoListTaiga.display_list_style1(Paperdink.epd, 123, 205, 4, 2, 125);
                             ^
Multiple libraries were found for "WiFi.h"
 Used: /home/hakayova/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /home/hakayova/Yazilim/arduino-1.8.19/libraries/WiFi
exit status 1
'class PaperdinkUIDateClass' has no member named 'display_calendar'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

hakova, 2023-01-08:
I worked a couple of hours for this today with frustrating results. First, I downloaded a new version of Paperdink library from [here](https://github.com/paperdink/PaperdInk-Library). I moved the old library from /home/user/Arduino/libraries to somewhere else, and put the new one in the same location naming it Paperdink. I realized in the process that /home/user/Arduino/libraries/Paperdink/src directory was missing the common.h file and copied the paperthink_common.h and renamed it to common.h, so the directory contained both files, since both of them were needed during compilation (dirty hack). I also downloaded the Arduino JSON library and added it to the project due to compilation errors and corrected a typo in some file, which I now forgot its name, form Arduino_Json to Arduino_JSON I belive. Despite all these efforts there are now new compilation errors related to fonts and WiFi I believe as seen below. 

Any guidance will be greatly appreciated.

```
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:5:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Roboto_Regular_12pt7b.h:163:7: error: 'GFXglyph' does not name a type
 const GFXglyph Roboto_Regular12pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Roboto_Regular_12pt7b.h:260:7: error: 'GFXfont' does not name a type
 const GFXfont Roboto_Regular12pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:6:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman9pt7b.h:85:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman9pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman9pt7b.h:182:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman9pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:7:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman15pt7b.h:217:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman15pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman15pt7b.h:314:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman15pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:8:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold15pt7b.h:230:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman_Bold15pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold15pt7b.h:327:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman_Bold15pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:9:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold10pt7b.h:115:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman_Bold10pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold10pt7b.h:212:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman_Bold10pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:10:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_Heavy25pt7b.h:763:7: error: 'GFXglyph' does not name a type
 const GFXglyph Mont_Heavy25pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_Heavy25pt7b.h:860:7: error: 'GFXfont' does not name a type
 const GFXfont Mont_Heavy25pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:11:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_ExtraLight8pt7b.h:70:7: error: 'GFXglyph' does not name a type
 const GFXglyph Mont_ExtraLight8pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_ExtraLight8pt7b.h:167:7: error: 'GFXfont' does not name a type
 const GFXfont Mont_ExtraLight8pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:12:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Gobold_Thin9pt7b.h:111:7: error: 'GFXglyph' does not name a type
 const GFXglyph Gobold_Thin9pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Gobold_Thin9pt7b.h:208:7: error: 'GFXfont' does not name a type
 const GFXfont Gobold_Thin9pt7b PROGMEM = {
[... 36 lines cut ...]
     Paperdink_Weather.font = &Gobold_Thin9pt7b;
                               ^
/home/user/Arduino/libraries/Paperdink/src/fonts.h:19:33: error: 'Code_New_Roman_Bold10pt7b' was not declared in this scope
 #define PAPERDINK_FONT_SML_BOLD Code_New_Roman_Bold10pt7b
                                 ^
[... cut ...]
/home/user/Arduino/user/calther/calther.ino:94:37: note: in expansion of macro 'PAPERDINK_FONT_SML'
     Paperdink_TodoListTaiga.font = &PAPERDINK_FONT_SML;
                                     ^
Multiple libraries were found for "WiFi.h"
 Used: /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /home/user/software/arduino-1.8.19/libraries/WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

rgujju, 2023-01-08:
We have pushed a new version of the library (v0.0.3). It should be available in Arduino library within 24 hours. You can get it from [here](https://github.com/paperdink/PaperdInk-Library/releases/tag/v0.0.3) earlier as well.
To manually install library from zip, remove any old libraries from the Arduino libraries folder and unzip the latest one.

closed 2023-01-08 by rgujju

hakova, 2023-01-09:
Thank you for pushing the new library version, but it did not solve any of the error messages. Moreover, there are several typos that need to be corrected to eventually get the error messages I am getting. Here is a non-exhaustive list of problems that need fixed to get to the stage of getting the error messages below:

- There is no common.h file in the library root. I copied the paperdink_common.h with the name of common.h under the same directory as a dirty hack
- There is a typo in paperdink_common.h (and therefore common.h as well): '#include ArduinoJson.h' should read '#include Arduino_JSON.h'

Then when you compile you get the exact same errors that I reported above:

```
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:5:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Roboto_Regular_12pt7b.h:163:7: error: 'GFXglyph' does not name a type
 const GFXglyph Roboto_Regular12pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Roboto_Regular_12pt7b.h:260:7: error: 'GFXfont' does not name a type
 const GFXfont Roboto_Regular12pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:6:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman9pt7b.h:85:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman9pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman9pt7b.h:182:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman9pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:7:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman15pt7b.h:217:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman15pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman15pt7b.h:314:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman15pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:8:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold15pt7b.h:230:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman_Bold15pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold15pt7b.h:327:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman_Bold15pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:9:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold10pt7b.h:115:7: error: 'GFXglyph' does not name a type
 const GFXglyph Code_New_Roman_Bold10pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Code_New_Roman_Bold10pt7b.h:212:7: error: 'GFXfont' does not name a type
 const GFXfont Code_New_Roman_Bold10pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:10:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_Heavy25pt7b.h:763:7: error: 'GFXglyph' does not name a type
 const GFXglyph Mont_Heavy25pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_Heavy25pt7b.h:860:7: error: 'GFXfont' does not name a type
 const GFXfont Mont_Heavy25pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:11:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_ExtraLight8pt7b.h:70:7: error: 'GFXglyph' does not name a type
 const GFXglyph Mont_ExtraLight8pt7bGlyphs[] PROGMEM = {
       ^
/home/user/Arduino/libraries/Paperdink/src/fonts/Mont_ExtraLight8pt7b.h:167:7: error: 'GFXfont' does not name a type
 const GFXfont Mont_ExtraLight8pt7b PROGMEM = {
       ^
In file included from /home/user/Arduino/libraries/Paperdink/src/fonts.h:12:0,
                 from /home/user/Arduino/user/calther/calther.ino:2:
/home/user/Arduino/libraries/Paperdink/src/fonts/Gobold_Thin9pt7b.h:111:7: error: 'GFXglyph' does not name a type
 const GFXglyph Gobold_Thin9pt7bGlyphs[] PROGMEM = {
[... 37 lines cut ...]
                               ^
In file included from /home/user/Arduino/user/calther/calther.ino:2:0:
/home/user/Arduino/libraries/Paperdink/src/fonts.h:19:33: error: 'Code_New_Roman_Bold10pt7b' was not declared in this scope
 #define PAPERDINK_FONT_SML_BOLD Code_New_Roman_Bold10pt7b
                                 ^
/home/user/Arduino/user/calther/calther.ino:93:42: note: in expansion of macro 'PAPERDINK_FONT_SML_BOLD'
     Paperdink_TodoListTaiga.font_bold = &PAPERDINK_FONT_SML_BOLD;
                                          ^
/home/user/Arduino/libraries/Paperdink/src/fonts.h:18:28: error: 'Code_New_Roman9pt7b' was not declared in this scope
 #define PAPERDINK_FONT_SML Code_New_Roman9pt7b
                            ^
[... cut ...] acro 'PAPERDINK_FONT_SML'
     Paperdink_TodoListTaiga.font = &PAPERDINK_FONT_SML;
                                     ^
Multiple libraries were found for "WiFi.h"
 Used: /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /home/user/software/arduino-1.8.19/libraries/WiFi
exit status 1
'Gobold_Thin9pt7b' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

rgujju, 2023-01-09:
@hakova Arduino_JSON and ArduinoJSON are two different libraries. Please install Arduino_JSON by Benoit Blanchon.
Also, looking at your error messages, Adafruit GFX library may be missing or an older version might be present. Please install the latest version.

emilyboda, 2023-01-09:
I downloaded v0.0.4 of the paperdink library and got an error message that there was no such file as ArduinoJson.h. I manually searched for and installed ArduinoJSON by Beniot Blanchon in the library manager and my code now successfully compiles. Not sure why I needed to do this manually when the description for the paperdink library says that the arduinojson library would be installed as well.

I still was having an issue where the wifi wouldn't connect. When I switched the wifi details from my 5GHz network to my 2.4GHz network I do not have any issues and the device connects to wifi.

It seems like the device doesn't work with 5GHz networks and that's something you should put in the instructions.

heytcass, 2023-01-09:
@rgujju I created two pull requests to correct the calender misspelling. @hakova Were there other misspellings you found?

hakova, 2023-01-10:
@heytcass No, I don't think so.