paperdink/PaperdInk-Library #2: Uploading wave example fails

dmikalova, 2022-12-29:
I am trying to set up the [wave example](https://github.com/paperdink/PaperdInk-Library/tree/main/examples/Wave). I set the variables in `config.h`, and am using the latest version of the library from the repo. It compiles fine, but I get an issue when uploading to my paperd classic - error below. Some other relevant information - I'm on an M1 Mac and have the board set to `ESP32 Dev Module` and the port set to the USB device for the paperd. The screen flashes on and off but after about 20 seconds it fails and then reboots.

```
Sketch uses 1048222 bytes (79%) of program storage space. Maximum is 1310720 bytes.
Global variables use 55888 bytes (17%) of dynamic memory, leaving 271792 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port /dev/cu.usbserial-01BFAC0E
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
Failed uploading: uploading error: exit status 2
```

haarts, 2022-12-30:
I ran into the same issue today. Also a M1 Mac. I installed the CP210x libs too.

haarts, 2022-12-30:
Ha! Fixed it. When you see 'Connecting .....' in the terminal output press the 'Reset' button on the back of the Paperd once. The upload will then commence.

haarts, 2022-12-30:
I'm unable to reproduce it! It worked precisely once.

stijnstijn, 2022-12-31:
You need to time it just right. My method is to start pressing the reset button just as the last `.` in a `........_____` series appears and to stop pressing it just as the first `_` appears. Usually it uploads fine then, sometimes it needs a few tries.

`esptool.py` does try to restart the ESP32 automatically, but it looks like the paperd.ink didn't set up the circuitry correctly for that method to work, so unfortunately the manual method is all that seems to work.

dmikalova, 2023-01-02:
I was able to get it! As the `...` appear I'll press, and then try to release right before the first `_` - by holding to a count of a quick 1-2-3. Each ..._ series is another attempt to try.

rgujju, 2023-01-09:
This seems to be an issue with macbook M1 devices.

See below similar issues:
https://github.com/espressif/esptool/issues/806
https://github.com/espressif/esptool/issues/712

There is a possible solution here which you can try:
https://github.com/espressif/esptool/issues/712#issuecomment-1003069098

adeshchopade, 2023-01-09:
I was able to resolve this issue by connecting my Paperd using a Type A to Type C cable via a Type C hub. 

I ran across the same issue on my MacBook Pro with M1 chipset. I was using a Type C to Type C cable when this error was occurring.

I have tried this multiple times now, I think it has something to do with the USB C drivers for Mac M1. On windows, USB C to USB C works fine.

haarts, 2023-01-09:
I can confirm the 'Type A to Type C cable via a Type C hub' solution works. This is one of the crazier things I've run into lately.

stijnstijn, 2023-01-09:
> @stijnstijn Paperd.Ink does implement the automatic reset circuitry correctly as it works fine on Windows and Linux systems. This problem seems to occur on macbook M1 devices.

I have this issue on both an M1 Mac and Windows 10. I've tried different versions of the USB-to-serial driver, different USB ports, and different cables and it still persists.

Would it be useful to open a separate issue about this, or is this really a general ESP32 problem that can't be addressed by the Paperd.ink team?

rgujju, 2023-01-10:
@stijnstijn Do you mean Windows 10 on an M1 Macbook or Windows 10 on x86?

radimkarnis, 2023-01-10:
@adeshchopade @haarts a possible solution for this was recently implemented in https://github.com/espressif/esptool/commit/353cefc85d7011d8d26d38861e8c736160df2ce6. Have you tried the most recent esptool? The latest development version containing this fix can be installed with `pip install esptool==4.5.dev1`.

stijnstijn, 2023-01-11:
> @stijnstijn Do you mean Windows 10 on an M1 Macbook or Windows 10 on x86?

Windows 10 on x86 (x64).

danielo515, 2023-02-09:
I can confirm the problem exists in my M1 mac. However,a s reported by @radimkarnis the latest esptool seems to work fine.
The steps I followed to fix it where:

1. Install esptool using a proper python version

```bash
python3 -m pip3 install esptool==4.5.dev1
```
2. Make arduino output verbose, so you can easily copy the "almost" correct command
[image]

3. Copy the failing command, because Arduino will use the esptool version they ship with (I hate this tool, seriously) and replace it with your proper esptool command. For example, instead of doing:
```bash
# Incorrect, not working, wrong, wrong 🛑 

/Users/your_user/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /Users/your_user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions/boot_app0.bin 0x1000 /Users/your_user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 /var/folders/8s/5v_sh3wx79d__9_75rbfxrxc0000gn/T/arduino_build_495/hello_world.ino.bin 0x8000 /var/folders/8s/5v_sh3wx79d__9_75rbfxrxc0000gn/T/arduino_build_495/hello_world.ino.partitions.bin
```
You will be doing this
```bash
python3 -m esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /Users/your_user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions/boot_app0.bin 0x1000 /Users/your_user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 /var/folders/8s/5v_sh3wx79d__9_75rbfxrxc0000gn/T/arduino_build_495/hello_world.ino.bin 0x8000 /var/folders/8s/5v_sh3wx79d__9_75rbfxrxc0000gn/T/arduino_build_495/hello_world.ino.partitions.bin
```

radimkarnis, 2023-02-09:
Great job @danielo515! Unfortunately, Arduino isn't very flexible and sometimes you have to resort to hacks like this... FIY, I wrote a documentation article about this process [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html).

danielo515, 2023-02-11:
> Great job @danielo515! Unfortunately, Arduino isn't very flexible and sometimes you have to resort to hacks like this... FIY, I wrote a documentation article about this process [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html).

Thanks for the link and the docs.
What I would love to see is a way to configure Arduino to use the right version. So far, my attempts of setting it to use the command that I manually use have failed. I tried editing the file `/Users/$USER/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/platform.txt` to use the same esptool that I do:
```
vim /Users/$USER/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/platform.txt
```
 but it always fails.