paperdink/PaperdInk-Library #4: Usage with PlatformIO

alhirzel, 2022-12-30:
Any information or a quickstart on using Paperd.ink with [platformio](https://platformio.org/)?

grdwyer, 2023-01-08:
This platform io config is working for me.
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	; paperdink/Paperdink@^0.0.2
	https://github.com/paperdink/PaperdInk-Library.git
	bblanchon/ArduinoJson@^6.20.0
	squix78/JsonStreamingParser@^1.0.5
monitor_speed = 115200
```
  
I got the Calther example running but had to make a couple changes. I kept getting connection issues for getting the weather info and needed to add `client->setInsecure();`  below [here](https://github.com/paperdink/PaperdInk-Library/blob/7b9d2ba4dc715afcdb7986028865aaad54819a2b/src/ui/weather/weather.cpp#L15)  

The calendar also wasn't having the right day offset which I ended up just hard coding day_offset [found here](https://github.com/paperdink/PaperdInk-Library/blob/7b9d2ba4dc715afcdb7986028865aaad54819a2b/src/ui/date_time/date_time.cpp#L41) to 6 (week starts on monday)

rgujju, 2023-01-08:
@grdwyer Thank you for the platform IO config. The rest of the issues you mention have been fixed.

closed 2023-01-08 by rgujju

alhirzel, 2023-01-08:
Thank you @grdwyer !

rgujju, 2023-01-08:
Yep! But please make sure to include all the dependent libraries:

    Adafruit GFX library by Adafruit
    GxEPD2 by Jean-Marc Zingg version
    JSON Streaming Parser by Daniel Eichhorn
    ArduinoJSON by Benoit Blanchon