paperdink/PaperdInk-Library #5: Crypto example does not compile

alhirzel, 2022-12-30:
I get the following error:

```
/home/alhirzel/.arduino15/packages/esp32/hardware/esp32/2.0.3/libraries/WiFi/src/WiFiGeneric.cpp: In static member function 'static bool WiFiGenericClass::setDualAntennaConfig(uint8_t, uint8_t, wifi_rx_ant_t, wifi_tx_ant_t)':
/home/alhirzel/.arduino15/packages/esp32/hardware/esp32/2.0.3/libraries/WiFi/src/WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
/home/alhirzel/.arduino15/packages/esp32/hardware/esp32/2.0.3/libraries/WiFi/src/WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/alhirzel/.arduino15/packages/esp32/hardware/esp32/2.0.3/libraries/WiFi/src/WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
exit status 1
```

Is there something I am missing?

haarts, 2022-12-30:
I did got it to compile. I basically followed the Getting started guide: https://docs.paperd.ink/docs/software/getting-started/.
Make sure you have the required libraries: https://docs.paperd.ink/docs/software/libraries/. Not to mention the 0.0.2 version of the paperd lib.

Please include what OS/library versions you are using.

rgujju, 2023-01-08:
@alhirzel Can you try with the latest version of ESP32 arduino core. ie v2.0.6

cimm, 2023-01-15:
I tested this today with the Paperdink 0.0.5 library and the esp32 version 1.0.6 board in the Arduino 2.0.3 IDE. The sketch compiles and runs on the Paperdink Classic without issues.

closed 2023-01-15 by rgujju