paperdink/PaperdInk-Library #7: Greyscale levels

grdwyer, 2023-01-08:
I am trying to display the grey levels on the display but it seems the Paperdink_Classic class uses the black and white display library [found here](https://github.com/paperdink/PaperdInk-Library/blob/7b9d2ba4dc715afcdb7986028865aaad54819a2b/src/devices/classic.h#L13).   
Could you provide a configuration for the device which would allow a minimal example like this to run?

```
#include <Arduino.h>
#include "config.h"
#include <Paperdink.h>

PAPERDINK_DEVICE Paperdink;

void setup()
{
  DEBUG.begin(115200);
  Paperdink.begin();
  Paperdink.enable_display();
  Paperdink.epd.fillScreen(GxEPD_WHITE);
  Paperdink.epd.fillRect(0, 0, 100, 100, GxEPD_BLACK);
  Paperdink.epd.fillRect(100, 0, 100, 100, GxEPD_DARKGREY);
  Paperdink.epd.fillRect(200, 0, 100, 100, GxEPD_LIGHTGREY);
  Paperdink.epd.fillRect(300, 0, 100, 100, GxEPD_WHITE);

  Paperdink.epd.display();
}

void loop(){

}

```

rgujju, 2023-01-14:
We have yet to integrate the 4 color library with this library. 
You can use this example https://github.com/paperdink/Paperdink_Classic_4C for now.

closed 2023-01-14 by rgujju

claaslange, 2023-11-07:
@rgujju when switching to that example we lose the quality of life features like button wake up from sleep etc, don't we? Any timeline on when the 4 color library will be integrated? The website is explicitly mentioning the ability to display 4 grayscale levels for the classic Paperdink.