paperdink/PaperdInk-Library #8: Calther example does not compile

hakova, 2023-01-09:
Please re-open [the closed issue](https://github.com/paperdink/PaperdInk-Library/issues/1) because the problem is not solved. None of the error messages were cleared by the new library (v0.0.4).

rgujju, 2023-01-09:
@hakova I think some library is missing in your setup. Have you installed the following libraries?

[image]
[image]
[image]
[image]
[image]
[image]

hakova, 2023-01-09:
Thank you @rgujju . I confirm that all of them are installed and I have to admit that a few more unnecessary libraries are installed during my desperate troubleshooting attempts (e.g. Arduino_JSON).

I re-installed the whole arduino 1.8.19 using the install.sh executable, this time with sudo privileges, thinking that it may have caused an incomplete installation last time around. After that, with library v0.0.4, I still cannot compile the darn thing. This time the error messages are different though:

```
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp: In member function 'int8_t PaperdinkUIWeatherClass::fetch_data(const char*, const char*, const char*, const char*)':
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:79:13: error: 'DynamicJsonDocument' was not declared in this scope
             DynamicJsonDocument json(35 * 1024);
             ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:85:13: error: 'DeserializationError' was not declared in this scope
             DeserializationError error = deserializeJson(json, https.getStream());
             ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:87:17: error: 'error' was not declared in this scope
             if (error) {
                 ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:93:17: error: 'JsonObject' was not declared in this scope
                 JsonObject root = json.as<JsonObject>();
                 ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:95:70: error: 'root' was not declared in this scope
                 snprintf(Paperdink_Weather.weather_string, 10, "%s", root["list"][0]["weather"][0]["main"].as<const char*>());
                                                                      ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:95:111: error: expected primary-expression before 'const'
                 snprintf(Paperdink_Weather.weather_string, 10, "%s", root["list"][0]["weather"][0]["main"].as<const char*>());
                                                                                                               ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:96:89: error: expected primary-expression before 'float'
                 Paperdink_Weather.day_max_temp = root["list"][0]["main"]["temp_max"].as<float>();
                                                                                         ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:97:89: error: expected primary-expression before 'float'
                 Paperdink_Weather.day_min_temp = root["list"][0]["main"]["temp_min"].as<float>();
                                                                                         ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:98:75: error: expected primary-expression before 'float'
                 Paperdink_Weather.rain = root["list"][0]["rain"]["1h"].as<float>();
                                                                           ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:99:75: error: expected primary-expression before 'float'
                 Paperdink_Weather.snow = root["list"][0]["snow"]["1h"].as<float>();
                                                                           ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:101:79: error: expected primary-expression before 'float'
                     Paperdink_Weather.rain = root["list"][0]["rain"]["3h"].as<float>();
                                                                               ^
/home/user/Arduino/libraries/Paperdink/src/ui/weather/weather.cpp:104:79: error: expected primary-expression before 'float'
                     Paperdink_Weather.snow = root["list"][0]["snow"]["3h"].as<float>();
                                                                               ^
Multiple libraries were found for "WiFi.h"
 Used: /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /home/user/software/arduino-1.8.19/libraries/WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Do you think it is due to some library collusion based on the error messages above? If so, how can I uninstall unneeded libraries?

hakova, 2023-01-10:
Alright, I deleted all directories created by arduino and re-installed it along with the libraries that it fails to install (Json Streamer and ArduinoJson), confirmed all other libraries were installed too and recompiled, and it worked. However, now I cannot upload it to the board. I am using linux and the message I get is pasted below. My understanding is that my user doesn't have permission to access/write to /dev/ttyS0. I tried /dev/ttyS1 as well with no avail. I will try this on a Windows machine next. I suspect that the VCP drivers won't be available for linux either.

```
Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Sketch uses 1030162 bytes (78%) of program storage space. Maximum is 1310720 bytes.
Global variables use 57672 bytes (17%) of dynamic memory, leaving 270008 bytes for local variables. Maximum is 327680 bytes.
Traceback (most recent call last):
esptool.py v3.0-dev
  File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open
Serial port /dev/ttyS0
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/ttyS0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 3682, in <module>
    _main()
  File "/home/user/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 3675, in _main
    main()
  File "/home/user/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 3329, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/home/user/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py", line 263, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/lib/python3.10/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyS0: [Errno 13] Permission denied: '/dev/ttyS0'
the selected serial port serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyS0: [Errno 13] Permission denied: '/dev/ttyS0'
 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

hakova, 2023-01-10:
Windows cannot communicate with the device either. To install the VCP driver, I right-clicked the silabser.inf file on the driver folder from silicon labs and clicked install. Windows asked permission for making changes on the computer, but otherwise there was no indication of anything installing, nor any confirmation of installation being complete. Windows device manager reports Silicon Labs CP201x USP to UART bridge at COM73. Uploading to this port (COM73) fails as seen below. During the communication attempt, device screen flashes several times and after the failure, the device resets. That actually makes me think that my device is perhaps broken.

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Sketch uses 1030162 bytes (78%) of program storage space. Maximum is 1310720 bytes.

Global variables use 57672 bytes (17%) of dynamic memory, leaving 270008 bytes for local variables. Maximum is 327680 bytes.

esptool.py v3.0-dev

Serial port COM73

Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch

_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

hakova, 2023-01-10:
I also realized that Windows does not recognize the device when the USB cable is plugged in, despite what is reported on Device Manager. I think I am done fiddling with this for a while. Good luck in your development. Please stop selling these devices until they are a little bit more ready for use.

closed 2023-01-14 by rgujju