paperdink/PaperdInk-Library #11: Unable to install library: "No valid dependencies solution found: dependency 'ArduinoJSON' is not available"

haarts, 2023-01-12:
I am trying to upgrade the library from 0.0.2 to 0.0.4. But it fails to install in the IDE.

On the console of the Arduino IDE I see:

`2023-01-12T19:49:23.290Z root ERROR Request listDependencies failed with error: 9 FAILED_PRECONDITION: No valid dependencies solution found: dependency 'ArduinoJSON' is not available`.

I believe the name of the dependency is wrong and should be `ArduinoJson`?

haarts, 2023-01-12:
I just tried and can confirm this is indeed the issue.

(Tried by cloning the repo, changing the `library.properties` and creating a zip of the entire lib and loading that into the Arduino IDE)

cimm, 2023-01-12:
I can confirm this fix works in Arduino 2.0.3 as tested in #13.

PR #12 (2023-01-14): Rename dependency ArduinoJSON to ArduinoJson
This naming mismatch prevented the library from installing in the Arduino IDE.

Fixes #11
closed 2023-01-14 by rgujju