paperdink/PaperdInk-Library #13: Quotes example does not compile

cimm, 2023-01-12:
I freshly installed Arduino 1.8.19 both from the [flatpack](https://flathub.org/apps/details/cc.arduino.arduinoide) repositories as directly from the Arduino [website](https://www.arduino.cc/en/software) on Fedora 36 as on macOS Ventura. I added the additional board URL as explained in the [documentation](https://docs.paperd.ink/docs/software/getting-started/) and choose the ESP32 Dev Module. This works fine.

When adding the Paperdink library the Arduino IDE asks if it should install all dependencies, which I accepted. During the library install I get the following error.

```
no protocol:
java.lang.RuntimeException: java.net.MalformedURLException: no protocol:
at
cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:250)
    at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.MalformedURLException: no protocol:
    at java.net.URL.<init>(URL.java:611)
    at java.net.URL.<init>(URL.java:508)
    at java.net.URL.<init>(URL.java:457)
    at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
    at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:56)
    at
cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:148)
    at
cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:118)
    at
cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:240)
    ... 1 more

Arduino: 1.8.19 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"
```

I restart the Arduino software and open the "Quotes" example because this seems like the simplest example and I just want to test if everything works. When I compile the sketch I get:

```
Multiple libraries were found for "WiFi.h"
Used: /Users/simon/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
Not used: /private/var/folders/wj/7dr02y8s1vx92scj6y7tknww0000gn/T/AppTranslocation/F313BE5E-D617-4A1B-9C46-1E8E29FCD9A9/d/Arduino.app/Contents/Java/libraries/WiFi
In file included from /Users/simon/Documents/Arduino/libraries/Paperdink/src/ui/crypto/crypto.h:5:0,
   from /Users/simon/Documents/Arduino/libraries/Paperdink/src/Paperdink.h:17,
    from /var/folders/wj/7dr02y8s1vx92scj6y7tknww0000gn/T/arduino_modified_sketch_72738/Quotes.ino:27:/Users/xxx/Documents/Arduino/libraries/Paperdink/src/ui/crypto/../../paperdink_common.h:5:25: fatal error: ArduinoJson.h: No such file or directorycompilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.
```

Note that it fails on the `crypto.h` header while I am trying to compile the "Quotes" example. I am not sure why the `Paperdink.h` header needs to include unrelated examples?

Any idea what I am doing wrong? Thank you!

haarts, 2023-01-12:
I can help with your Arduino 2.0.3 issue. I just ran into that too and fixed it. See: #11

haarts, 2023-01-12:
I've just tried it in the 2.x IDE and the quotes example works. Albeit with the fixed library as described in #11 and the associated pull request.

cimm, 2023-01-12:
Thank you @haarts. I manually installed the Paperdink Arduino library with your fix in pull request #11 and it does indeed compile and I can write it to the Paperdink device. Success! Let's hope the pull request gets merged soon.

closed 2023-01-14 by rgujju

cimm, 2023-01-15:
Pull request #11 has been merged and a new version of the Paperdink library, version 0.0.5, has been released.

I tested the Quotes example with the new version of the library on both Arduino 1.8.19 and 2.0.3 and the example compiles correctly on both versions. The "no protocol" and "multiple libraries" errors have both been resolved with the fix from #11.

This issue has been resolved.