paperdink/PaperdInk-Library #20: Read the battery status (and show a nice graphic)?

noonasGit, 2023-01-16:
Hello!

I *love* my little merlot device - getting me back in to coding (and learning a few new tricks too)..

I see the wifi part but could not see any references to battery (charging status, battery level)
Is there a place where I can poke this to make a nice battery graphic for my Papered.ink screen?

/jonas

timendum, 2023-02-02:
I found something in [this repository](https://github.com/paperdink/timcal2/tree/main/timcal2), but it's archived. I also agree, a function to gather and display the battery status in the library would be very useful.

haarts, 2023-02-05:
[This](https://github.com/paperdink/timcal2/blob/main/timcal2/timcal2.ino#L118-L128) seems the relevant part for the battery. But it uses a deprecated function `adc_power_on`, that should be `adc_power_acquire`.
It also includes stuff that makes no sense to me like `adc1_config_width(ADC_WIDTH_BIT_12);` or `pcf8574.digitalWrite(BATT_EN, HIGH);`. There are also some magic constants that are unclear to me: `(batt_adc/4095.0)*4.2)`.

I would love to know what that all means.

elratt0r, 2023-05-11:
See here:
https://github.com/paperdink/paperd_ink_test/pull/4

danielo515, 2024-07-01:
Thanks for the link. It is a bit basic though, it just has an example. Is there any docs or example with a UI part?